ARM-software/arm-systemready #2: Build failure when run the command: ./build-scripts/build-ir-live-image.sh

eslover, 2021-05-24:
There is one build break when run the command: 
./build-scripts/build-ir-live-image.sh. The failure as the followings. 

~/workspace/arm-systemready/IR/scripts ~/workspace/arm-systemready/IR/scripts ~/workspace/arm-systemready/IR/scripts/edk2-test/uefi-sct ~/workspace/arm-systemready/IR/scripts/edk2-test ~/workspace/arm-systemready/IR/scripts ~/workspace/arm-systemready/IR/scripts ~/workspace/arm-systemready/IR/scripts
./build-scripts/build-linux.sh: line 98: outpath: invalid indirect expansion
Build failed: error while running source at line 106 in ./build-scripts/build-linux.sh for [][].

I'm using ubuntu 20.04 on x86_64 machine.

r64343@lava2:~$ cat /etc/issue
Ubuntu 20.04.2 LTS \n \l

Jason Liu

Jiss-Jose, 2021-05-25:
Hi Jason Liu,  

Thanks for raising the issue.  

We have added support  for build on 20.04 system and updated IR Readme  for the same.
Could you please  check it from your side ?

Regards
Jiss Jose

eslover, 2021-05-26:
Hi Jiss Jose

The issue resolved with your fix. Thanks. Now, the build is okay with Ubuntu 20.04. 

The operation has completed successfully.
Completed preparation of disk image for busybox boot
----------------------------------------------------

closed 2021-05-26 by edhay