ARM-software/arm-systemready #64: System IR guide does not work properly!

Jingzhao123, 2023-03-07:
Hi,
I am doing a test for system-IR. Based on the latest version(commit: 032e50e91ecfe2256d8e6ddd5099def0015a9f2f, Mar 6), I follow the guideline of https://github.com/ARM-software/arm-systemready/tree/main/IR. But the command in the guideline does not work properly:

```
$ qemu-system-aarch64 -bios ${path2flash}/nor_flash.bin -drive file=${path2sysIR}/ir_acs_live_image.img,if=virtio,format=raw  -cpu cortex-a57 -smp 2 -m 2048 -M virt,secure=on -monitor null -no-acpi -nodefaults -nographic -rtc base=utc,clock=host -serial stdio -d unimp,guest_errors -machine virtualization=on
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.8(release):v2.8-493-g2b7150b38
NOTICE:  BL1: Built : 16:23:05, Mar  7 2023
NOTICE:  BL1: Booting BL2
NOTICE:  BL2: v2.8(release):v2.8-493-g2b7150b38
NOTICE:  BL2: Built : 16:23:06, Mar  7 2023
NOTICE:  BL1: Booting BL31
NOTICE:  BL31: v2.8(release):v2.8-493-g2b7150b38
NOTICE:  BL31: Built : 16:23:09, Mar  7 2023

U-Boot 2023.04-rc3-00082-gb08ffdffdc (Mar 07 2023 - 16:22:10 +0800)

DRAM:  2 GiB
Core:  51 devices, 14 uclasses, devicetree: board
Flash: 32 MiB
Loading Environment from Flash... *** Warning - bad CRC, using default environment

In:    pl011@9000000
Out:   pl011@9000000
Err:   pl011@9000000
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Scanning for bootflows in all bootdevs
Seq  Method       State   Uclass    Part  Name                      Filename
---  -----------  ------  --------  ----  ------------------------  ----------------
Scanning global bootmeth 'efi_mgr':
Scanning bootdev 'fw-cfg@9020000.bootdev':
fatal: no kernel available
No working controllers found
scanning bus for devices...
Unknown seq -1 for label 'scsi'
Scanning bootdev 'virtio-blk#32.bootdev':
** Bad device specification virtio 20 **
  0  efi          ready   virtio       1  virtio-blk#32.bootdev.par efi/boot/bootaa64.efi
** Booting bootflow 'virtio-blk#32.bootdev.part_1' with efi
No EFI system partition
No EFI system partition
Failed to persist EFI variables
Missing TPMv2 device for EFI_TCG_PROTOCOL
Missing RNG device for EFI_RNG_PROTOCOL
No UEFI binary known at beeccc30
Boot failed (err=-22)
Unknown seq -1 for label 'nvme'
No ethernet found.
No more bootdevs
---  -----------  ------  --------  ----  ------------------------  ----------------
(1 bootflow, 1 valid)
=>
```

Qemu version:
```
$ qemu-system-aarch64 --version
QEMU emulator version 4.2.1 (Debian 1:4.2-3ubuntu6.24)
Copyright (c) 2003-2019 Fabrice Bellard and the QEMU Project developers
```
Is there anything wrong with my operation?

edhay, 2023-03-13:
Hi @Jingzhao123 ,

 The SystemReady IR ACS image build system has moved to new Yocto based build. Could you please follow the new README here: 
https://github.com/ARM-software/arm-systemready/blob/main/IR/Yocto/README.md

The old README, that you had followed shall be de-commissioned soon.
Please feel free to contact us for any further queries.

SystemReady ACS Support

Jingzhao123, 2023-03-14:
> 

Hi, edhay,
Based on the yocto guide, there are still some errors!

```
Initialising tasks: 100% |################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 16 Local 0 Mirrors 0 Missed 16 Current 1149 (0% match, 98% complete)
NOTE: Executing Tasks
WARNING: Duplicate inclusion for /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/../poky/meta/recipes-kernel/linux/linux-yocto.inc in /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/../recipes-kernel/linux/linux-yocto_6.0.bb
ERROR: bsa-acs-drv-1.0-r0 do_compile: ExecutionError('/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/temp/run.do_compile.2140639', 2, None, None)
ERROR: Logfile of failure stored in: /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/temp/log.do_compile.2140639
Log data follows:
| DEBUG: Executing shell function do_compile
| ==================================================================
| This script will compile the Linux BSA Kernel module
| ==================================================================
| echo "Kernel source is set to /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source"
| Kernel source is set to /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source
| make -C /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source M=/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/platform/pal_linux/files modules
| make[1]: Entering directory '/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source'
| make[2]: Entering directory '/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-build-artifacts'
|   CC [M]  /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/platform/pal_linux/files/./src/pal_pe.o
| /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/platform/pal_linux/files/./src/pal_pe.c:111:1: error: conflicting types for 'pal_pe_call_smc'; have 'void(ARM_SMC_ARGS *)'
|   111 | pal_pe_call_smc(ARM_SMC_ARGS *ArmSmcArgs)
|       | ^~~~~~~~~~~~~~~
| In file included from /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/platform/pal_linux/files/include/pal_linux.h:32,
|                  from /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/platform/pal_linux/files/./src/pal_pe.c:22:
| /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/platform/pal_linux/files/../../../val/include/pal_interface.h:157:6: note: previous declaration of 'pal_pe_call_smc' with type 'void(ARM_SMC_ARGS *, int32_t)' {aka 'void(ARM_SMC_ARGS *, int)'}
|   157 | void pal_pe_call_smc(ARM_SMC_ARGS *args, int32_t conduit);
|       |      ^~~~~~~~~~~~~~~
| make[3]: *** [/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source/scripts/Makefile.build:249: /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/platform/pal_linux/files/./src/pal_pe.o] Error 1
| make[2]: Leaving directory '/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-build-artifacts'
| make[2]: *** [/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source/Makefile:1852: /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/platform/pal_linux/files] Error 2
| make[1]: *** [Makefile:222: __sub-make] Error 2
| make[1]: Leaving directory '/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source'
| make: *** [Makefile:42: all] Error 2
| cp: cannot stat 'bsa_acs_pal.o': No such file or directory
| echo "Kernel source is set to /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source"
| Kernel source is set to /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source
| make -C /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work-shared/generic-arm64/kernel-source M=/home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/bsa-acs-drv/1.0-r0/linux-acs/bsa-acs-drv/files/val modules
[... cut ...]
NOTE: Tasks Summary: Attempted 2973 tasks of which 2969 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/jingzhao/coding/system_IR/arm-systemready/IR/Yocto/meta-woden/build/../recipes-acs/bsa-acs-drv/bsa-acs-drv.bb:do_compile
Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.
2023-03-14 12:59:29 - ERROR    - Command returned non-zero exit status 1
+ '[' 1 -eq 0 ']'
+ popd
~/coding/system_IR/arm-systemready/IR/Yocto
```

Jingzhao123, 2023-03-14:
On the other hand, I can not find the commit(032e50e91ec, Mar 6) in the latest arm-systemready repo. Is it missing?

edhay, 2023-03-14:
Hi @Jingzhao123, 

    The build issue above is due to a compilation issue in the Linux BSA code, a dependent software. The same is now fixed.
Now please try again the same steps again. 
https://github.com/ARM-software/arm-systemready/blob/main/IR/Yocto/README.md#acs-build-steps

SystemReady ACS Support

freeuser56, 2023-03-20:
Hi @edhay,

I got the same error with prebuilt image v22.10_2.0.0_BETA-1. How to fix it, thanks.

NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.8(release):v2.8-528-gff65ac244
NOTICE:  BL1: Built : 02:41:30, Mar 14 2023
NOTICE:  BL1: Booting BL2
NOTICE:  BL2: v2.8(release):v2.8-528-gff65ac244
NOTICE:  BL2: Built : 02:41:32, Mar 14 2023
NOTICE:  BL1: Booting BL31
NOTICE:  BL31: v2.8(release):v2.8-528-gff65ac244
NOTICE:  BL31: Built : 02:41:35, Mar 14 2023

U-Boot 2023.04-rc4 (Mar 14 2023 - 02:40:14 +0000)

DRAM:  2 GiB
Core:  51 devices, 14 uclasses, devicetree: board
Flash: 32 MiB
Loading Environment from Flash... *** Warning - bad CRC, using default environment

In:    pl011@9000000
Out:   pl011@9000000
Err:   pl011@9000000
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
Scanning for bootflows in all bootdevs
Seq  Method       State   Uclass    Part  Name                      Filename
---  -----------  ------  --------  ----  ------------------------  ----------------
Scanning global bootmeth 'efi_mgr':
Scanning bootdev 'fw-cfg@9020000.bootdev':
fatal: no kernel available
No working controllers found
scanning bus for devices...
Unknown seq -1 for label 'scsi'
Scanning bootdev 'virtio-blk#32.bootdev':
** Bad device specification virtio 20 **
  0  efi          ready   virtio       1  virtio-blk#32.bootdev.par efi/boot/bootaa64.efi
** Booting bootflow 'virtio-blk#32.bootdev.part_1' with efi
No EFI system partition
No EFI system partition
Failed to persist EFI variables
Missing TPMv2 device for EFI_TCG_PROTOCOL
Missing RNG device for EFI_RNG_PROTOCOL
No UEFI binary known at beec5a70
Boot failed (err=-22)
Unknown seq -1 for label 'nvme'
No ethernet found.
No more bootdevs
---  -----------  ------  --------  ----  ------------------------  ----------------
(1 bootflow, 1 valid)
=>

chetan-rathore, 2023-03-20:
Hi @Jingzhao123, 

The issue seems to be related to u-boot. At our side with below u-boot version, the image is booting properly and grub option is seen.

U-Boot 2021.10-rc1-00232-ga0953b34d9

NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.5(release):v2.5-284-g87311b4c1-dirty
NOTICE:  BL1: Built : 23:15:55, Aug  4 2021
NOTICE:  BL1: Booting BL2
NOTICE:  BL2: v2.5(release):v2.5-284-g87311b4c1-dirty
NOTICE:  BL2: Built : 23:15:56, Aug  4 2021
NOTICE:  BL1: Booting BL31
NOTICE:  BL31: v2.5(release):v2.5-284-g87311b4c1-dirty
NOTICE:  BL31: Built : 21:25:37, Aug 16 2021

**U-Boot 2021.10-rc1-00232-ga0953b34d9 (Mar 08 2023 - 18:35:30 +0530)**

DRAM:  2 GiB
Flash: 32 MiB
Loading Environment from Flash... *** Warning - bad CRC, using default environment

In:    pl011@9000000
Out:   pl011@9000000
Err:   pl011@9000000
Net:   No ethernet found.
Hit any key to stop autoboot:  0
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
scanning bus for devices...

Device 0: unknown device

Device 0: 1af4 VirtIO Block Device
            Type: Hard Disk
            Capacity: 524.0 MB = 0.5 GB (1073186 x 512)
... is now current device
Scanning virtio 0:1...
load - load binary file from a filesystem

Usage:
load <interface> [<dev[:part]> [<addr> [<filename> [bytes [pos]]]]]
    - Load binary file 'filename' from partition 'part' on device
       type 'interface' instance 'dev' to address 'addr' in memory.
      'bytes' gives the size to load in bytes.
      If 'bytes' is 0 or omitted, the file is read until the end.
      'pos' gives the file byte position to start reading from.
      If 'pos' is 0 or omitted, the file is read from the start.
Scanning disk virtio-blk#32...
Found 4 disks
Missing RNG device for EFI_RNG_PROTOCOL
No EFI system partition
BootOrder not defined
EFI boot manager: Cannot load any image
Found EFI removable media binary efi/boot/bootaa64.efi
910848 bytes read in 16 ms (54.3 MiB/s)
Booting /efi\boot\bootaa64.efi
Welcome to GRUB!

error: no such device: ((hd0,gpt1)/efi/boot)/EFI/BOOT/grub.cfg.
error: serial port `com0' isn't found.
 
Can you please share the steps used to build the nor_flash.bin at your side. Also if possible, can you attach your nor_flash.bin here.

Thanks,
ACS team

freeuser56, 2023-03-21:
Hi @edhay 

I compiled according to the steps in the [readme](https://github.com/ARM-software/arm-systemready/blob/main/IR/Yocto/README.md#to-build-the-firmware-image-follow-the-below-steps)
--------
mkdir working_directory
cd working_directory
repo init -u https://github.com/glikely/u-boot-manifest
repo sync
export CROSS_COMPILE=<path to gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/bin/aarch64-none-elf->
make qemu_arm64_defconfig
make
----------
[nor_flash.bin](https://drive.google.com/file/d/1bm_fitvzXbUdjOdFoi_24AuXFDBpkcSx/view?usp=share_link)

According to the BBR documentation, the EFI_RNG_PROTOCOL is required.
Is there a problem with my QEMU environment?
**Missing RNG device for EFI_RNG_PROTOCOL**

Thanks

chetan-rathore, 2023-03-21:
Hi @freeuser56, 

I was able to reproduce the issue at my end with the nor_flash.bin, thanks for sharing the bin. 

It seems the issue is due to some change in how u-boot scans virtio drive and loads "bootaa64.efi".  We are debugging this.
Please run below commands on u-boot terminal manually to proceed in meantime.

=> virtio scan
=> virtio info
=> fatload virtio 0 $kernel_addr_r EFI/BOOT/bootaa64.efi
=> bootefi $kernel_addr_r $fdtcontroladdr

Successful run of these commands should give grub screen, in case still no grub screen is seen. Please share the output of these commands.

Thanks,
ACS team

freeuser56, 2023-03-21:
Hi @chetan-rathore 

Successfully running on the u-boot terminal manually.

Thanks

chetan-rathore, 2023-03-21:
Some updates:

It seems there is some change in latest u-boot boot cmd flow, which has caused this.  We are trying to get the last commit with which image was booting fine and will update README to use that tag/commit.

Jingzhao123, 2023-03-21:
@chetan-rathore 
Thanks for your reply.
Now, I use the prebuild image, but not the manual build image temporarily. And please let me know if it has been fixed. 
Thanks again!