ARM-software/arm-systemready #69: Can rajgoy01 use shorter paths?

hrw, 2023-03-29:
Many lines in log files have path to source file where message was located:

MediaIoProtocols:/home/rajgoy01/WORK/sbsa/sr_oob/SR/scripts/edk2-test/uefi-sct/SctPkg/TestCase/UEFI/EFI/Generic/SbbrRequiredUefiProtocols/BlackBoxTest/RequiredUefiProtocolsBBTestFunction.c:126:EFI_LOAD_FILE_PROTOCOL

Can those paths be shorter? Use some CI instead of rajgoy01 account?

rajatgoyal47, 2023-03-30:
Hello Marcin,

Thank you for your suggestion. Please note that the path after "edk2-test" is not entirely under our control as it is determined by SctPkg.
We will be implementing changes that involve using the Common CI account for pre-built binaries. This will help reduce the length of the first half of the path leading to the source file (~3 Directory less).

Thanks,
Rajat

hrw, 2023-03-30:
```/opt/uefi-sct/SctPkg/TestCase/UEFI/EFI/Generic/SbbrRequiredUefiProtocols/BlackBoxTest``` feels shorter

edhay, 2023-06-13:
Hi Marcin, 

Achieving a path like for edk2-test
/opt/uefi-sct/SctPkg/TestCase/UEFI/EFI/Generic/SbbrRequiredUefiProtocols/BlackBoxTest
will bring in special handling in only some scripts in the ACS build system which we want to avoid.

Is the below path generated by CI acceptable?
/home/CIUSER/ACSv2.0.0/SR/scripts/edk2-test/uefi-sct/SctPkg/TestCase/UEFI/EFI/Generic/SbbrRequiredUefiProtocols/BlackBoxTest/RequiredUefiProtocolsBBTestFunction.c

This will be shorter than the example in the first message which is a non-standard path.

hrw, 2023-06-13:
I am fine with whatever you make as it can always be removed from log.

edhay, 2023-06-14:
Thank you for the reply, Marcin.

chetan-rathore, 2023-10-05:
Hi @hrw,

With the recent CI infrastructure changes of using GitHub actions for building images, no internal paths or names are logged as part of ACS run. 

The new path is :  /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2-test/uefi-sct/
SctPkg/TestCase/UEFI/EFI/Generic/EfiCompliant/BlackBoxTest/EfiCompliantBBTestReq
uired_uefi.c
 
Please note that due to the GitHub actions setup, arm-systemready will come twice.

Thanks,
ACS team

closed 2023-10-09 by chetan-rathore