ARM-software/arm-systemready #81: pass rate for SR certification

am675, 2023-06-10:
Hi experts,

I hope you are doing great. I am currently working on a project related to the SystemReady
[SR_bbr_acs_log_caselist.txt](https://github.com/ARM-software/arm-systemready/files/11711794/SR_bbr_acs_log_caselist.txt)
 SR certification and I have run into some issues that I need assistance with.

Specifically, I tested the certification on a qemu [sbsa-ref](https://gitlab.arm.com/systemready/firmware-build/qemu/sbsa-ref-ve-es) device and found that the test results were not fully passed. There were about 32 failed test cases and 113 cases that were not supported out of a total of 1000 test cases. Detail test log attached. I am wondering if there is a guide or any resources available that can help me analyze the failed test cases? Also, is there a guide that can explain which specific cases can cause the certification to fail, and which ones won't impact the outcome of the certification?

I appreciate any guidance or resources you can provide to help me better understand how I can 
[SR_bbr_acs.log](https://github.com/ARM-software/arm-systemready/files/11711793/SR_bbr_acs.log)
improve the certification results. Thanks for taking the time to read this email.

Best regards,
Yin

edhay, 2023-06-13:
Hi Yin,

    Thank you for the question:
There are no specific guidelines explaining which specification test case can fail.
The overall process for certifications is provided in the document: https://developer.arm.com/documentation/den0109/latest Appendix B
In summary: Partner runs the ACS on the System under tests and submits the logs. 
Ideally we expect all tests to pass. However there may be failures.
During the certification process, SystemReady Certification team from Arm will analyze the failures and may grant waivers if the same does not cause impact on OS boot or normal functioning of the system (This is only a general guideline). If not, they will request the partner to solve the issues and re-test with their updated firmware.
On case to case basis other actions may be advised. 

For analysis of failed test case, a local debug has to be done. The case of UEFI-SCT failures, the analysis to be done with firmware and UEFI-SCT suite. 

Please do reply for any further queries.

SystemReady ACS Team

cc: @sunnywang-arm

am675, 2023-06-13:
Hi Edhaya,
Thanks for you quick response, will get you if any further queries.

closed 2023-06-13 by am675