ARM-software/arm-systemready #90: After running the sbsa tool in v23.01_2.0.0_BETA-0 , system assert

zhangyongping8, 2023-08-04:
https://github.com/ARM-software/arm-systemready/tree/main/SR 

In  release details pre-built_image v23.01_2.0.0_BETA-0 (updated on Jan 16 ), with sbsa.efi version 7.1.0 

After running this sbsa.efi in  UEFI efi2 , system assert! 

"ASSERT [ArmCpuDxe] /data/aone/space/224612335/source/workspace/uefi/ArmPkg/Library/DefaultExceptionHandlerLib/AArch64/DefaultExceptionHandler.c(333): ((BOOLEAN)(0==1))

zhangyongping8, 2023-08-04:
or when will the latest version of the completed build be updated

chetan-rathore, 2023-08-04:
Hello @zhangyongping8,

Can you please also share the terminal or Sbsa logs to know what caused the exception/assert.

The latest version of the pre-built image will be available around Sep23-end, but the latest SR image can be built easily at your end by following the build steps mentioned here (https://github.com/ARM-software/arm-systemready/tree/main/SR#steps-to-build-systemready-sr-acs-live-image-1)

Let us know if any issues are faced while building the image.

Thanks,
ACS team

chetan-rathore, 2023-08-23:
Hi @zhangyongping8,

Any update on the logs, and also can you let us know if the issue is seen on a consistent basis.

Thanks,
ACS team

chetan-rathore, 2023-08-31:
Hi @zhangyongping8,

We are closing this issue, Please reopen this if issue still observed on your side.

Thanks,
ACS team

closed 2023-08-31 by chetan-rathore