ARM-software/arm-systemready #189: Add acpica-tools to ACS Linux

sunnywang-arm, 2024-05-23:
so that ACS Linux can collect ACPI dump logs by running the comments below. 

acpidump > acpi.log
acpixtract -a acpi.log
iasl -d *.dat

By the way, the goal is to move https://gitlab.arm.com/systemready/systemready-es-sr-template/-/blob/main/os-logs/linux-distro-cmds.sh to https://github.com/ARM-software/arm-systemready to have single script file used in ACS test and OS test.

chetan-rathore, 2024-06-19:
Hello @sunnywang-arm, 

We are working on this ticket, will add below config to SR ACS buildroot.
config BR2_PACKAGE_ACPICA

Thanks,
ACS team

closed 2024-06-25 by chetan-rathore

sunnywang-arm, 2024-06-25:
@chetan-rathore  Could you also get someone to run the commands below and then uncomment the lines in init.sh?
```
 # acpidump > /mnt/acs_results/linux_dump/acpi.log
 # acpixtract -a > /mnt/acs_results/linux_dump/acpi.log
 # iasl -d /mnt/acs_results/linux_dump/*.dat
```
 
https://github.com/sunnywang-arm/arm-systemready/blob/52d9f771f3d1b8bff7b207c363d85269a3a0198f/common/ramdisk/init.sh#L131

PR #195 (2024-06-25): Update buildroot_defconfig : Add acpica buildroot defconfig
Fixes #189 

Add support for acpica tool in ES and SR buildroot defconfig