ARM-software/arm-systemready #191: [AmpereAltraMax] No prompt available in graphic mode but serial console

anncheng-arm, 2024-06-12:
### Failure/Symptom
No prompt available after booting into ACS via graphic mode but working fine in serial console.
Unplug/plug external USB device, the USB messages were output properly in graphic mode/serial mode which means vga driver should work properly, looks more like init script/bootargs issues.

Booting logs from serial console: [acs_console.zip](https://github.com/user-attachments/files/15801099/acs_console.zip)

[image]

### Configuration
- System: SuperMicro Ampere Mt Hamilton ARS-520M-NRL
- ACS Version:
  - [Daily Build](https://github.com/ARM-software/arm-systemready/actions/runs/9470196117)
  - SR v23.09
  - SR v24.03

anncheng-arm, 2024-06-13:
I cannot find root.sh in entire file system which is used by initab, is this set correctly?

https://github.com/ARM-software/arm-systemready/blob/fd9e41c9551372ef0ed749963a17ca5c897e2dff/common/ramdisk/inittab#L33

chetan-rathore, 2024-06-19:
Hello @anncheng-arm, 

Omitting console=ttys0 and tty=AMA0 from the command line worked on rpi4, can you please check if it works on the DUT.

Thanks,
ACS team

anncheng-arm, 2024-06-19:
@chetan-rathore 

Here is Linux Kernel document to explain how /dev/console binds to getty (tty0 – graphic mode, ttyS0 – Serial console for 16550, ttyAMA0 – ARM UART PL011).
[serial-console](https://www.kernel.org/doc/html/latest/admin-guide/serial-console.html)

I've also tested this on my RPi4 and Ampere system. Removing ttyS0 and ttyAMA0 results in /dev/console binding to tty0, which is why you can see output and prompts on the graphical interface. However, this also means there is no output or prompt from the serial port anymore. You can simply move tty0 to be the last parameter, and the output/prompt will then go to the graphical mode.

I reported this issue because ODMs typically check from the graphical interface first. If there is no prompt, they might think the system has hung during testing. However, I am guessing if ACS is designed intentionally to direct all tests and prompts to the serial console only, it prevents accidental resets or test terminations from the graphical interface(tty0).

We may not need to address this issue if this design is intentional. Or we may just need to init the getty tty0 properly in /etc/inittab to get it to prompt which is good enough.

r1chard-lyu, 2025-08-07:
Hi all,
It's good to see that someone has identified this issue. I encountered the same issue while testing ACS on an Ampere server using iLO6. The prompts are not visible on the graphical interface. Following the suggestion in [the comment](https://github.com/ARM-software/arm-systemready/issues/191#issuecomment-2179225712), redirecting the console output resolves the issue.

I hope this can be properly fixed in the prebuilt image, as running server compliance tests remotely is very common.