ARM-software/arm-systemready #192: SystemReady IR ACS v2.1.1 tests are terminated with an error

davidhuziji, 2024-06-19:
Hi,

Recently I bumped IR ACS tests from v2.1.0 to **_v2.1.1_**. I just replaced previous prebuilt image with v24.03_2.1.1. Could you please tell me if any other component should be updated as well?
I ask because, ACS v2.1.1 is terminated with an error on my system while v2.1.0 runs well. I'm also a bit confused with the error log.
Do you mind taking a look at the logs below? Sorry they could be a bit long. I'd appreciate it if you can give me some suggestions on debugging the failure.

In my environment with ACS v2.1.1, UEFI SCT runs well until the last test case `UpdatePackageListFunction`. It reports timeout.
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 482, in read_nonblocking
    raise TIMEOUT('Timeout exceeded.')
pexpect.exceptions.TIMEOUT: Timeout exceeded.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File ".../lib/oeqa/runtime/cases/arm_systemready_ir_acs.py", line 22, in test_acs
    match_id = self.target.expect(self.console, test_patterns,
  File "xxx/../layers/meta-arm/meta-arm/lib/oeqa/controllers/fvp.py", line 145, in call_pexpect
    result = attr(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 340, in expect
    return self.expect_list(compiled_pattern_list,
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 119, in expect_loop
    return self.timeout(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 82, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7fe1e3d23340>
command: .../tmp_systemready-glibc/hosttools/telnet
args: ['.../tmp_systemready-glibc/hosttools/telnet', 'localhost', '5009']
buffer (last 100 chars): b'PE 4KB Granule Support          \r\r\n       PSCI_CPU_ON: failure[-2]\r\r\n       PSCI_CPU_ON: failure[-2]'
before (last 100 chars): b'PE 4KB Granule Support          \r\r\n       PSCI_CPU_ON: failure[-2]\r\r\n       PSCI_CPU_ON: failure[-2]'
after: <class 'pexpect.exceptions.TIMEOUT'>
```
But it still reports SCT completes and starts BSA test suite as usual. However, after BSA tests are running for a while, it will repeat the following steps (SCT completion message, lots of asserts and some BSA test cases) for several times.
```
WARNING: arm-systemready-ir-acs-2.1.1-r0 do_testimage: Could not load artifacts list, skip artifacts retrieval
Timed out waiting for login prompt.
Boot log follows:

[0;37;40mPress any key within 5 seconds 
Press any key within 4 seconds 
Press any key within 3 seconds 
Press any key within 2 seconds 
Press any key within 1 seconds 
SCT has completed run. Press any key to start SCT execution from the beginning. WARNING: Ensure you have backed up the existing logs.

[0;37;40mPress any key within 5 seconds 
Press any key within 4 seconds 
Press any key within 3 seconds 
Press any key within 2 seconds 
Press any key within 1 seconds 
Running CapsuleApp 

[0;37;40m

ASSERT_EFI_ERROR (Status = Not Found)

ASSERT [CapsuleApp] /home/runner/work/arm-systemready/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/uefi-apps/1.0-r0/edk2/MdeModulePkg/Library/UefiHiiServicesLib/UefiHiiServicesLib.c(94): !(((INTN)(RETURN_STATUS)(Status)) < 0)

... # Lots of ASSERTs

ASSERT [CapsuleApp] /home/runner/work/arm-systemready/arm-systemready/IR/Yocto/meta-woden/build/tmp/work/generic_arm64-oe-linux/uefi-apps/1.0-r0/edk2/MdePkg/Library/DxeHobLib/HobLib.c(49): mHobList != ((void *) 0)

Starting UEFI Debug dump

[0;37;40m[1;37;40mConnect[0;37;40m No drivers could be connected.

[0;37;40mPress any key to start BSA in verbose mode.

[0;37;40mIf no key is pressed then BSA will be run in normal mode

[0;37;40mPress any key within 10 seconds 
Press any key within 9 seconds 
Press any key within 8 seconds 
Press any key within 7 seconds 
Press any key within 6 seconds 
Press any key within 5 seconds 
Press any key within 4 seconds 
Press any key within 3 seconds 
Press any key within 2 seconds 
Press any key within 1 seconds 

 BSA Architecture Compliance Suite

          Version 1.0.8

 Starting tests with print level :  3

 Creating Platform Information Tables

  PE_INFO: Number of PE detected       :   16
  GIC CPUIF not present
   ...
  Peripheral: Num of UART controllers  :    1

      *** Starting PE tests ***

Operating System View:

   1 : Check Arch symmetry across PE         

       PSCI_CPU_ON: failure[-2]
       ...
       Checkpoint -- 290                           : Result:  FAIL

   2 : Check for number of PE                     : Result:  PASS

   3 : Check for AdvSIMD and FP support      

       PSCI_CPU_ON: failure[-2]
       ...
       PSCI_CPU_ON: failure[-2]
```
Eventually it reports an error and terminates the whole test.
```
WARNING: arm-systemready-ir-acs-2.1.1-r0 do_testimage: Can not get target disk usage: Failed to start FVP.
Transitioned to off
RESULTS:
RESULTS - arm_systemready_ir_acs.SystemReadyACSTest.test_acs: ERROR (72571.22s)
SUMMARY:
arm-systemready-ir-acs () - Ran 1 test in 72571.222s
arm-systemready-ir-acs - FAIL - Required tests failed (successes=0, skipped=0, failures=0, errors=1)
```

Thank you.

davidhuziji, 2024-06-19:
More detailed logs during the repeated part are shown below. As this log shows, although time-out is reported above, the SCT test summary still claims that it completes well.

```
------------------------------------------------------------

  Remaining test cases: 0

  Protocol test: UpdatePackageListFunction

  Instances: 1/1

  Iterations: 1/1

------------------------------------------------------------

Arm ACS Version: SystemReady IR ACS v2.1.1

BBR ACS 1.0.5 (EBBR)

UpdatePackageListFunction

Revision 0x00010000

Test Entry Point GUID: 00C3A3A0-1A1E-4609-9B0F-F2835301432E

Test Support Library GUIDs: 

  1F9C2AE7-F147-4D19-A5E8-255AD005EB3E

------------------------------------------------------------

UEFI 2.6

Test Configuration #0

------------------------------------------------------------

Function auto test for HII Database Protocol UpdatePackageList().

------------------------------------------------------------

Logfile: "\EFI\BOOT\bbr\SCT\Log\HIITest\HIIDatabaseProtocolTest0\UpdatePackageListFunction_0_0_00C3A3A0-1A1E-4609-9B0F-F2835301432E.log"

Test Started: 06/15/24  08:04p

------------------------------------------------------------

Returned Status Code: Success

UpdatePackageListFunction: [NOT SUPPORTED]

  Passes........... 0

  Warnings......... 0

  Errors........... 0

------------------------------------------------------------

UEFI 2.6

Revision 0x00010000

Test Entry Point GUID: 00C3A3A0-1A1E-4609-9B0F-F2835301432E

------------------------------------------------------------

Logfile: "\EFI\BOOT\bbr\SCT\Log\HIITest\HIIDatabaseProtocolTest0\UpdatePackageListFunction_0_0_00C3A3A0-1A1E-4609-9B0F-F2835301432E.log"

Test Finished: 06/15/24  08:04p
[... 152 lines cut ...]

 Creating Platform Information Tables

 PE_INFO: Number of PE detected       :   16
 ...
 Peripheral: Num of UART controllers  :    1

      *** Starting PE tests ***

Operating System View:

   1 : Check Arch symmetry across PE         

       PSCI_CPU_ON: failure[-2]
       ...
       Checkpoint -- 290                           : Result:  FAIL

   2 : Check for number of PE                     : Result:  PASS

   3 : Check for AdvSIMD and FP support      

       PSCI_CPU_ON: failure[-2]
       ...
       PSCI_CPU_ON: failure[-2]Connection closed by foreign host.
```

chetan-rathore, 2024-06-19:
Hi @davidhuziji, 

We will look into the issue and will keep you updated on the findings.
Can you also try the latest IR build image on the system and share observations.

Image Link: https://github.com/ARM-software/arm-systemready/actions/runs/9537746488/artifacts/1606480747

Thanks,
ACS team

davidhuziji, 2024-06-19:
Thanks a lot @chetan-rathore . I will try and update you.

davidhuziji, 2024-06-24:
Hi @chetan-rathore, I tried this new prebuilt image and it worked as expected.
Thank you so much.

May I know if you have a plan to update v24.03_2.1.1 prebuilt image in this public repo?

chetan-rathore, 2024-06-25:
Hi @davidhuziji,

We have updated the stable build.

Thanks,
ACS team

closed 2024-06-25 by chetan-rathore

davidhuziji, 2024-06-25:
Hi @chetan-rathore,

Thank you so much.
May I know how to track the prebuilt image changes? Currently I can only rely on the SHA values to distinguish different image versions. Is any tags or minor versions available for reference? Thank you.

PR #196 (2024-06-25): Updated stable build image
Fixes #192