ARM-software/arm-systemready #234: security-interface-extension-keys missing from pre-built image

mwasilew, 2024-12-03:
The instructions for BBSR call for setting up secure boot with keys from pre-built image. The keys should be located in `security-interface-extension-keys` directory: https://developer.arm.com/documentation/107981/0200/SystemReady-compliance-testing---BBSR-and-Security-Interface-Testing-FAQ/How-do-I-test-for-Base-Boot-Security-Requirements--BBSR--compliance-/Enroll-the-Secure-Boot-keys?lang=en

This directory is missing from the latest pre-built image for DeviceTree band:
https://github.com/ARM-software/arm-systemready/tree/main/SystemReady-devicetree-band/prebuilt_images/v24.11_3.0.0-BET0

The directory is available in the pre-built IR image from March: 
https://github.com/ARM-software/arm-systemready/tree/main/IR/prebuilt_images/v24.03_2.1.1

Please update either the pre-built image (preferred) or the documentation.

amrathesh, 2024-12-04:
Hi @mwasilew, 

Thank you for raising this. There were a few directory restructurings with the recent release, and the BBSR keys were moved to `<bootfs>/acs_tests/bbsr-keys/`. Please use this new location. We will work on updating the user guide in the meantime.

Cheers,
Amrathesh
System Compliance Team

mwasilew, 2024-12-05:
@amrathesh Thanks for the fix. From my pov this ticket can be closed now.

closed 2024-12-05 by chetan-rathore