ARM-software/arm-systemready #243: Hitting this: "Error-Input address is not aligned. Masking the last 2 bits" when using the v24.11_3.0.0-BET0 ACS image.

nvidiaarm, 2024-12-10:
I tried running both the SystemReady band ACS and SCT for BBSR suites using the v24.11_3.0.0-BET0 ACS image.
Both testing is hitting this error during the run: Error-Input address is not aligned. Masking the last 2 bits

amrathesh, 2024-12-12:
Hi @nvidiaarm, 

Can you please provide more information on this, at what point of ACS execution, or which test are you observing this error. Please attach corresponding logs.

Best Regards,
Amrathesh
Arm System Compliance Team.

nvidiaarm, 2024-12-13:
Hi,

This error occurred during the SystemReady band ACS and SCT for BBSR runs using the ACS live image - https://github.com/ARM-software/arm-systemready/tree/main/SystemReady-band/prebuilt_images/v24.11_3.0.0-BET0. Both testing is hitting this: Error-Input address is not aligned. Masking the last 2 bits after running for 2 hours into the testing.

***@***.***

Thank you.

amrathesh, 2024-12-16:
Hi @nvidiaarm,

Upon reviewing the error message, it appears that the issue may stem from the SBSA/BSA ACS suite execution and unexpected to encounter such an error during BBSR SCT. Further analysis suggests that the error might be triggered by an unaligned address encountered by pal_mmio_read[1]. (unaligned address might be referenced in system's ACPI info) Please provide log snippets to further debug and root cause the issue.

[1] [ARM-software/bsa-a/pal/uefi_acpi/common/src/pal_misc.c#L159](https://github.com/ARM-software/bsa-acs/blob/de0064098150651bb186c01cda74aef1b0923121/pal/uefi_acpi/common/src/pal_misc.c#L159)

Best Regards,
Amrathesh
Arm System Compliance Team.

nvidiaarm, 2024-12-19:
See the attached. Thanks.

amrathesh, 2024-12-24:
Hi @nvidiaarm, 

I don't see any attachments in previous mail, it might have dropped, can you please resend, you can also directly mail the logs to systemready-support@arm.com.

Best Regards,
Amrathesh
Arm System Compliance Team.

closed 2025-01-31 by amrathesh