ARM-software/arm-systemready #276: systemready-dt_acs_live_image.wic.xz.zip v24.11_SR_REL3.0.0-BETA0_SR-DT_REL3.0.0-BETA0 image doesn't work on TI SK-AM62P-LP boar

jetm, 2025-03-06:
Installing the `systemready-dt_acs_live_image.wic` image from the TI board using the Arago Yocto OS produces a corrupt result. The image content can be seen from other Linux OSes and different architectures, but there is something that is not OK from the image: the board is unable to read the content.

The workaround is recreating the image again, and the image can be installed in the TI board.

jetm, 2025-03-06:
@sam-moss-arm Hi Sam, hope you're the right Sam to tag for this issue found in Linaro

chetan-rathore, 2025-03-10:
Hi @jetm,

Can you share more details on the issue ? 
* How you have installed the pre-built image on TI board, using a USB driver or SD card
* What is going corrupt, and if the test are not running or results are getting corrupt.

Some logs will be helpful for us for debugging.

Thanks,
Chetan

jetm, 2025-03-10:
I think we sorted it out. Arago image mounts all the eMMC partitions by default, and `dd` **must** use unmounted. Closing for now. Thanks

closed 2025-03-10 by jetm