ARM-software/arm-systemready #306: ACS 3.0 some of jason is showing no waive

quic-pansing, 2025-03-20:
Picture attached.
As per guideline BSA failure will not impact on IR certification. but ACS 3.0 jaison mention that no waiver. 

[image]

quic-pansing, 2025-04-01:
@chetan-rathore we checked with latest image, here is below issue we found

[image]

Second issue is:  when apply waiver, on logs is showing waiver applied but Dashboard there is no waiver shown 

[image]

chetan-rathore, 2025-04-03:
Hello @quic-pansing,

Thanks for running the latest image and sharing the feedbacks. 

1)  SCT is a mandatory test suite for overall compliance, and not waivable.  So waivers will not be applied on SCT.
2)  Do you mean acs_summary**.xml**  or acs_summary**.json**,  as currently support for XML is not provided and JSON is the standard format.
3) BSA is recommended for DT compliance and hence not impact overall compliance.
4) BBSR is also recommended for now.

Thanks,
Chetan

quic-pansing, 2025-04-03:
@chetan-rathore SCT Is mandatory but if some particular test case is failed due to Limitation or other reason, how waiver can apply?

chetan-rathore, 2025-04-03:
@quic-pansing ..can you share more details on which sct test failed and the limitation...you can send email to support-systemready-acs@arm.com with details

quic-pansing, 2025-04-04:
sure. Will share over email..

chetan-rathore, 2025-04-14:
Hi @quic-pansing,

Thanks for sharing the results, we are discussing internally the results.

chetan-rathore, 2025-04-24:
> [@chetan-rathore](https://github.com/chetan-rathore) any update for above. When we can expect answer

Hi @quic-pansing,

We are discussing internally and will share response once have the updates.

Thanks,
Chetan

quic-pansing, 2025-05-09:
Hey @chetan-rathore, 
   did you get any response?

ownia, 2025-06-05:
Hi @chetan-rathore, could you share the FAQ guide with me?

chetan-rathore, 2025-06-06:
Hi @ownia, 

I also noticed it, but please note as pivot from certification to self guided compliance is for both bands most of non ES/SR information is valid for both bands.

But I will sync internally to have FAQ guide more clearer.

quic-pansing, 2025-06-06:
> Hi [@quic-pansing](https://github.com/quic-pansing),
> 
> I have shared the response and the systemready compliance FAQ guide which will help to understand the method of declaration of compliance after pivot to self guided compliance.
> 
> Thanks, Chetan

Network protocol related failure
[PlatformSpecificElements_0_0_A0A8BED3-3D6F-4AD8-907A-84D52EE1543B.log](https://github.com/user-attachments/files/20622660/PlatformSpecificElements_0_0_A0A8BED3-3D6F-4AD8-907A-84D52EE1543B.log)

Query mode related failure logs
[QueryMode_Func_0_0_D493E3FD-6B56-4F8E-BA45-606D0E8F5671.log](https://github.com/user-attachments/files/20622662/QueryMode_Func_0_0_D493E3FD-6B56-4F8E-BA45-606D0E8F5671.log)

chetan-rathore, 2025-06-18:
Hi @quic-pansing, 

Please find the responses from our side.

* UEFI Compliant failure on HII:
This is a false positive due to recent U-Boot now implementing HiiConfigRouting and the SCT parser config not being up to date. **This can be ignored.**

* UEFI Compliant failure on network boot:
**This is a true failure: EFI_SIMPLE_NETWORK_PROTOCOL is missing (SNP - No).**

* QueryMode failure on mode 0:
This looks like a true failure, is this a special setup with console on both UART & GOP maybe?
Can you share details of your setup.

Thanks,
Chetan

psingh3086, 2025-07-16:
@chetan-rathore did you get any update?

chetan-rathore, 2025-07-16:
Hi @psingh3086,

As the SystemReady program has pivoted to a self-guided compliance model, partners are now expected to review the root cause of the failure and document any applicable waivers internally. The ACS image will not auto-apply the waivers for failures in mandatory test cases.

For the failures related to ACS suite, I will be sharing a SystemReady DT intermittent image, which includes fixes for known false failures due to tooling or test issues within the ACS. This should help ensure accurate results and reduce noise in your compliance testing.

Thanks,
Chetan

chetan-rathore, 2025-08-09:
Hi @psingh3086, 

Can you run with this latest debug image and share the observations. 

https://github.com/chetan-rathore/arm-systemready/tree/v3.0.2/SystemReady-devicetree-band/prebuilt_images/v3.0.1plus-rc0

If the acs_results folder can be shared, it will be great help.

Thanks,
Chetan

smangali-qc, 2025-09-24:
HI @chetan-rathore ,
we are still observing same failure with the above image. getting stuck  501  test

**logs:**
 407 : Check System Counter Frequency        
       START  Using F/W interrupt API's

       Counter frequency is 19 MHz
       B_TIME_02
                                       : Result:  PASS
       END

      One or more Timer tests failed or were skipped.

      *** Starting Wakeup semantic tests ***

Operating System View:
 501 : Wake from EL1 PHY Timer Int           
       START  Using F/W interrupt API's
  Using F/W interrupt API's
  Using F/W interrupt API's

       PE wakeup by some other event

Regards,
shiva kumar

chetan-rathore, 2025-09-25:
Thanks for quick update, will have a look