ARM-software/arm-systemready #330: DT self test

quic-pansing, 2025-04-03:
Hi Arm,
   is DT self test mandatory or can be apply as Waiver?

chetan-rathore, 2025-04-03:
Hello @quic-pansing, 

The DT Kselftest is recommended and not mandatory. It will not impact the overall compliance result.
Waiver can be applied for DT Kselftest.

Will advise to use the latest code in this branch https://github.com/ARM-software/arm-systemready/tree/3.0_development for running.

Thanks,
Chetan

quic-pansing, 2025-04-04:
Thanks @chetan-rathore . 
is MVP tests also recommending or mandatory?

chetan-rathore, 2025-04-04:
Hi @quic-pansing,

Eth tool, DT validate (schema) and Block devices are mandatory.
DT Kselftest (node probe) is recommended.

It will be good to run BBSR test also.

quicAspratap, 2025-04-07:
@chetan-rathore ,

The DT Kselftest (node probe) test is fails for below remoteproc subdevices because remoteproc driver calls the probe of these subdevices only when a DSP FW is started successfully. As arm-system ready image does not has QCOM specific DSP FW BINs installed under /lib/firmware, these subdevices do not probe. Is fine to a get a waiver for probe failure of such subdevice nodes? Can you please confirm so that we can raise a waiver request for these failures?

subdevices defined under remoteproc nodes:

--
/soc@0/remoteproc@3700000/glink-edge/fastrpc/compute-cb@3
/soc@0/remoteproc@3700000/glink-edge/fastrpc/compute-cb@4
/soc@0/remoteproc@3700000/glink-edge/fastrpc/compute-cb@5
/soc@0/remoteproc@a300000/glink-edge/fastrpc
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@1
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@11
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@12
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@13
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@14
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@2
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@3
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@4
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@5
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@6
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@7
/soc@0/remoteproc@a300000/glink-edge/fastrpc/compute-cb@8
/soc@0/remoteproc@3700000/glink-edge/apr/service@3
/soc@0/remoteproc@3700000/glink-edge/apr/service@4
/soc@0/remoteproc@3700000/glink-edge/apr/service@4/clock-controller
/soc@0/remoteproc@3700000/glink-edge/apr/service@4/dais
/soc@0/remoteproc@3700000/glink-edge/apr/service@7
/soc@0/remoteproc@3700000/glink-edge/apr/service@7/dais
/soc@0/remoteproc@3700000/glink-edge/apr/service@8
/soc@0/remoteproc@3700000/glink-edge/apr/service@8/routing
--

thanks,
Shivendra

quicAspratap, 2025-04-08:
Hi @chetan-rathore, 

Not sure on how to apply waiver. I was going by your comment in - https://github.com/ARM-software/arm-systemready/issues/330#issuecomment-2775572789. "Waiver can be applied for DT Kselftest." Will need you help on the process as-well.

thanks,
Shivendra

chetan-rathore, 2025-04-14:
Hi @quicAspratap, 

For the DT Kselftest, the ACS waiver process will allow waivers to be applied as its a recommended test suite.
@Ashishsharma1903 can provide a example waiver.json for DT kselftest.

thanks,
Chetan

Ashishsharma1903, 2025-04-17:
Hi @quicAspratap ,

I am attaching a sample or example waiver.json with which you can get a understanding of how waivers are applied, there are example waivers for dtkself test and dtvalidate in the json under "Suite": "Standalone_tests".

[waiver.json](https://github.com/user-attachments/files/19791738/waiver.json)

also here is the link for waiver guide which can help you to understand the process in a more efficient way:
[https://github.com/ARM-software/arm-systemready/blob/main/common/docs/waiver_guide.md](url)

let us know if you have any further queries, we will be happy to help and clear them.

Thanks,
Ashish Sharma

quic-pansing, 2025-04-17:
@chetan-rathore  @Ashishsharma1903  Kselftest is recommended, still waiver required?

ricardosalveti, 2025-04-22:
> arm-system ready image does not has QCOM specific DSP FW

Are the firmware files available via linux-firmware or from common places in oe-core? I don't see why we shouldn't be able to propose adding firmware as part of the image (can be rejected, but that is a different conversation). Did we validate if the tests are fine after adding the firmware packages?

quicAspratap, 2025-04-23:
> Are the firmware files available via linux-firmware or from common places in oe-core? I don't see why we shouldn't be able to propose adding firmware as part of the image (can be rejected, but that [...]

We did validate that DSP are able to start ( After manually pushing the matching FW BINs from META and then manually starting the DSP from userspace). Now the test case starts at boot-up so it fails the first iteration. Need to check if the test case can be re-run after the DSP has been manually started.

Shiva to confirm.

chetan-rathore, 2025-04-24:
> [@chetan-rathore](https://github.com/chetan-rathore) [@Ashishsharma1903](https://github.com/Ashishsharma1903) Kselftest is recommended not mandatory, still waiver required?

Hello @quic-pansing and @quicAspratap,

Waiver is specific to failures, for the recommended ones the ACS waiver processing flow will allow to apply waivers.
So if you want a clean report with waiver reason for recommended test, you can use the waiver processing.

For the mandatory ones, the waivers will not be applied and the report will capture the failures.

Based on your query on #306 ,  we are analyzing the case where you are requesting waiver  for a mandatory test suite and will reply on the 306 ticket.

Thanks,
Chetan

closed 2025-06-06 by quic-pansing