ARM-software/arm-systemready #402: CONFIG_SCSI_UFS_RENESAS not letting boot to shell

quic-pansing, 2025-06-05:
With v25.04_3.0.1, kernel 6.12 is not reaching the shell. Logs are attached.

After more trial and validation, we found that if we only disable CONFIG_SCSI_UFS_RENESAS , Device reach to shell. Our current device do not use ufs-renasas.c .

[logs_612.txt](https://github.com/user-attachments/files/20608568/logs_612.txt)

chetan-rathore, 2025-06-09:
Hi @quic-pansing, 

We have made changes to eliminate the xbsa linux test boot time dependency. Its possible even with this change for your system the CONFIG_SCSI_UFS_RENESAS config needs to be disabled.

Can you run the DT image at this location: https://github.com/chetan-rathore/arm-systemready/tree/Images/images/linux_porting and share your feedback on 
1) Any issues during Linux booting
2) Any crash around xBSA linux during dma table creation or running of 605 test  ref #252 

Thanks,
Chetan

quic-pansing, 2025-06-16:
Thanks @chetan-rathore 
   we checked image.at your given link there is 2 image 
1. DT image .wic  --->  Boot fine. Device reach to shell.

   
2.bootuptest .img ---> device not reaching to shell. Logs attached  

[putty_not_working_kernel.log](https://github.com/user-attachments/files/20753820/putty_not_working_kernel.log)

if possible, please share what is difference both images.

chetan-rathore, 2025-06-16:
Hi @quic-pansing, 

Thanks for confirming. The other image is for SystemReady band. 

SystemReady Compliance supports two bands systemReady band and Devicetree band. You can check more details at this page: https://www.arm.com/architecture/system-architectures/systemready-compliance-program, which details about the SystemReady program and the bands.

Thanks,
Chetan

Pradeep-pvk, 2025-06-18:
Hi Chetan,

Thanks for the details.
Can you please provide more details on the changes made for DT image.wic ?
we were curious to know how it got fixed (#402).

Thanks,
Pradeep

chetan-rathore, 2025-06-18:
Hi @Pradeep-pvk, 

The SystemReady image includes various test tools, one of which is the BSA test suite. Most BSA tests are implemented as UEFI applications, while certain tests—particularly those related to MSI and DMA functionality—are delivered as Linux kernel modules.

Historically, the BSA kernel module included a patch that introduced a dependency during Linux boot to retrieve DMA devices and their associated IOMMU groups. This patch was legacy code that dated back to the early development of SystemReady.

Recently, as part of a separate effort, we removed that legacy dependency (https://gitlab.arm.com/linux-arm/linux-acs/-/merge_requests/21) . The updated implementation now uses standard Linux APIs to acquire the required information, eliminating the need for early boot-time probing.

Based on logs shared by @quic-pansing in this ticket and #252, the crash dump shows failures occurring in the BSA API responsible for DMA information access during early boot. However, since the legacy patch has now been removed, this no longer affects Linux boot behavior.

Thanks,
Chetan

chetan-rathore, 2025-06-20:
Hi @quic-pansing, 

Would it be possible to close either #402 or #252? We can retain one of them for any follow-up questions or support.

closed 2025-06-20 by quic-pansing

Pradeep-pvk, 2025-06-23:
Hi Chetan,

which error log that was attached in this ticket(#402) confirms that it was a BSA issue ?
logs attached in this ticket was "With v25.04_3.0.1+ BSA Patch reverted".

Thanks and Regards,
Pradeep