ARM-software/arm-systemready #422: capsule update issue  (ported from bsa-acs)

chetan-rathore, 2025-06-25:
During the capsule update test for ARM System Ready, we observed that the script keeps looping around unauth.bin since it is not deleted after the test. This causes the script to never proceed to tampered.bin and signed_capsule.bin.

We suggest adding a flag or mechanism to indicate that once the test for unauth.bin is complete (pass or fail), the script should move forward to the next capsule.

if exist FS%r:\acs_tests\app\unauth.bin then
FS%r:\acs_tests\app\CapsuleApp.efi FS%r:\acs_tests\app\unauth.bin >> FS%r:\acs_results_template\fw\capsule-update.log
else
echo "unauth.bin not present" >> FS%r:\acs_results_template\fw\capsule-update.log
endif
     echo "entering tampered story"
echo "Testing tampered.bin update" >> FS%r:\acs_results_template\fw\capsule-update.log
echo "Test_Info" >> FS%r:\acs_results_template\fw\capsule-update.log
if exist FS%r:\acs_tests\app\tampered.bin then
FS%r:\acs_tests\app\CapsuleApp.efi FS%r:\acs_tests\app\tampered.bin >> FS%r:\acs_results_template\fw\capsule-update.log
else
echo "tampered.bin not present" >> FS%r:\acs_results_template\fw\capsule-update.log
endif

cc @quic-pansing

chetan-rathore, 2025-06-25:
Hi @quic-pansig

Can you please share details on @sathishas89  query.

Hi,
I am unable to reproduce the issue which you are facing.
Could you please share your logs for the same issue, and could you provide details of tested platform?
Also, can you confirm that whether system is resetting after execution of unauth.bin?

sathishas89, 2025-07-14:
Hi @psingh3086,
Any update on this issue?

psingh3086, 2025-07-14:
We are discussing internally. checking which logs we can provide, since logs have only run command info and device reset.

chetan-rathore, 2025-07-26:
Hi @smangali-qc and @psingh3086, 

As there is a new ticket #430  for capsule update issue with more details, can this ticket be closed.

Thanks,
Chetan

chetan-rathore, 2025-08-09:
Hi @smangali-qc and @psingh3086,

Closing this ticket, as issue/query related to capsule update is followed on #430  

Thanks,
Chetan

closed 2025-08-09 by chetan-rathore