ARM-software/arm-systemready #439: SystemReady BBSR-SCT Failures due to U-Boot UEFI Implementation Gap

asuvath17, 2025-08-07:
Hi @chetan-rathore,

We are encountering multiple BBSR-SCT test failures on our U-Boot-based platform during SystemReady certification. These failures are related to UEFI variables and services that are implemented in EDK2 but **not available in standard U-Boot**.

**Failed Tests:**

-  Secure Boot Variables: PKDefault, KEKDefault, dbDefault, dbxDefault (not found)
- Variable Services: BBSRVariableSizeTest - QueryVariableInfo() missing
- Security: EFI Image Execution Info Table missing
- TCG MOR: MemoryOverwriteRequestControl variables missing

**Cause :**  U-Boot lacks all these comprehensive UEFI variable ecosystem that EDK2 provides by default.

**Questions**

1.  Are these failures waivable for U-Boot platforms?
2. Is there any other certification path for U-Boot-based SystemReady platforms?

Guidance on whether these failures can be waived or if there are alternative approaches to achieve certification with U-Boot would be helpful.

Thanks,
Asuvath

dineshkumar-v85, 2025-08-12:
Hi, 
Any update on this query.

chetan-rathore, 2025-09-16:
Hi @dineshkumar-v85 and @asuvath17, 

As discussed we have made changes to separate the extension results from the SRS compliance results. If the system is not supporting BBSR requirements and the BBSR tests are skipped, they will not be shown as non-compliant but captured as not-run.

Please run with this image and share feedback: https://github.com/chetan-rathore/arm-systemready/tree/Images/SystemReady-devicetree-band/prebuilt_images/v3.1.0-rc0-12Sep

Thanks,
Chetan

sathishas89, 2025-10-30:
Hi @asuvath17 ,
Any update?

asuvath17, 2025-10-30:
Hi @chetan-rathore and @sathishas89 ,

Thank you for the update. I’ve tested the changes, and the behavior is as described .The skipped BBSR tests are now shown as "not run". The issue is resolved.

Best Regards,
Asuvath

closed 2025-10-30 by asuvath17