ARM-software/arm-systemready #442: DT_VALIDATE Reported as Non-Waived Failure Despite No Unwaived Failures

asuvath17, 2025-08-11:
In our ACS results summary for SystemReady DeviceTree band, DT_VALIDATE is listed under “non-waived fails in suite(s): DT_VALIDATE,” but there are no unwaived failures in either dt_kselftest or dt_validate. All failures in these suites are either waived or do not exist.

[image]

[image]

Please help us to resolve or interpret this result.

Attaching our logs for your reference.

[acs_summary.zip](https://github.com/user-attachments/files/21708866/acs_summary.zip)

ownia, 2025-08-19:
```
Traceback (most recent call last):
  File "/usr/bin/dt-validate", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 154, in main
    sg.check_dtb(filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 91, in check_dtb
    self.check_subtree(dt, subtree, False, "/", "/", filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 84, in check_subtree
    self.check_subtree(tree, value, disabled, name, fullname + name, filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 84, in check_subtree
    self.check_subtree(tree, value, disabled, name, fullname + name, filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 84, in check_subtree
    self.check_subtree(tree, value, disabled, name, fullname + name, filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 79, in check_subtree
    self.check_node(tree, subtree, disabled, nodename, fullname, filename)
  File "/usr/lib/python3.12/site-packages/dtschema/dtb_validate.py", line 34, in check_node
    for error in self.validator.iter_errors(node, filter=match_schema_file,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/dtschema/validator.py", line 441, in iter_errors
    for error in self.DtValidator(schema,
                 ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/validators.py", line 371, in iter_errors
    for error in errors:
                 ^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/_legacy_keywords.py", line 423, in unevaluatedProperties_draft2019
    evaluated_keys = find_evaluated_property_keys_by_schema(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/_legacy_keywords.py", line 399, in find_evaluated_property_keys_by_schema
    evaluated_keys += find_evaluated_property_keys_by_schema(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/jsonschema/_legacy_keywords.py", line 342, in find_evaluated_property_keys_by_schema
    resolved = validator._resolver.lookup(ref)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/referencing/_core.py", line 671, in lookup
    raise exceptions.Unresolvable(ref=ref) from error
```

chetan-rathore, 2025-09-16:
Hi @asuvath17, 

The DT-validate issue was due to incompatibility between in dt-schema version and default json-schema build with Yocto. We have made changes to build compatible json-schema. 
Please use this image to run at your end and share the results: https://github.com/chetan-rathore/arm-systemready/tree/Images/SystemReady-devicetree-band/prebuilt_images/v3.1.0-rc0-12Sep

Thanks,
Chetan