ARM-software/arm-systemready #458: Common Question on SR-DT

smangali-qc, 2025-09-05:
HI @chetan-rathore ,

1. **Boot Flow: BBSR-LINUX after BBSR-UEFI**
Once BBSR-UEFI completes, the system attempts to transition to BBSR-LINUX with UEFI Secure Boot enabled. However, the DTB authentication fails. We’d like clarification on the following:

          - Is bootaa64.efi signed?
          - If yes, we need the corresponding keys included in the pre-built image to sign the DTB.
          - If no, what is the recommended method to run BBSR-LINUX?

We tried clearing the keys, but that causes BBSR-UEFI to trigger again.
2. **Linux Distribution Testing**
We found the following ARM documentation for Linux distro testing:
🔗 https://developer.arm.com/documentation/DUI1101/4-0/Test-SystemReady-Devicetree-band/Test-installation-of-Linux-distributions?lang=en

We’d like to confirm:

Are we required to test only the distros listed in the ARM documentation?
Can we use the daily ISO builds from these sources instead?
🔗 https://download.opensuse.org/ports/aarch64/tumbleweed/iso/
🔗 https://get.opensuse.org/leap/15.6/

Regards,
Shiva Kumar

chetan-rathore, 2025-09-18:
Hi @smangali-qc, 

For the Linux Distribution Testing questions, 

[SRS ](https://developer.arm.com/documentation/den0109/latest/)sheds more light.

`The recommended distributions are Fedora, Debian, Ubuntu, RHEL, Rocky Linux, SLES, openSUSE, OpenWrt, and Yocto.
When selecting Linux distributions or BSDs, maximize coverage by installing an OS from different groups in the following list. Avoid repetition within groups unless all four groups are covered:
•
RHEL, Fedora, or Rocky Linux
•
SLES or openSUSE
•
Ubuntu or Debian
•
OpenWrt or Yocto` 

Daily ISO builds from source can be used.

Thanks,
Chetan

chetan-rathore, 2025-09-24:
Hi @sathishas89, 

Can you provide more details on the first part of query around the BBSR testing part.

sathishas89, 2025-11-07:
Hi @Shiva Kumar,

We can only ship public keys as part of the ACS package. Including private keys is not recommended due to significant security risks.

If private keys were distributed, there’s a possibility that a partner might accidentally leave the ACS public keys installed in a production system. In such a case, if the corresponding private keys are also available (e.g., from the ACS image), an attacker could use them to sign arbitrary firmware or EFI binaries. This would effectively bypass Secure Boot, compromising the integrity of the platform.

This type of issue has occurred in the industry before, where test or development keys were unintentionally left enabled in released products, leading to serious security vulnerabilities. Therefore, per Arm SystemReady guidelines and general Secure Boot best practices, only public keys (such as .der and .auth files) should be provided in the ACS image for validation and test purposes.

Could you please confirm whether your team is explicitly using a DTB (Device Tree Blob) during boot, or if the platform firmware (e.g., UEFI) is providing the device tree dynamically through the EFI configuration table?
This information will help us assess whether the DTB handling process requires any adjustments to align with the BBSR secure boot flow.

Best regards,
Sathisha S

smangali-qc, 2025-11-07:
HI @sathishas,

yes,  UEFI install the OS-DTB in  EFI configuration table  and provide it to kernel.  we have to explicitly sign the DTB with the keys which used for EFI  image . 

if keys are not part of prebuilt, we cannot sign the DTB and subsequently   device will fail to boot to shell. 

Regards,
Shiva Kumar

smangali-qc, 2025-11-10:
HI @sathishas,

I understand the security and compliance guidelines. Could you please suggest a straightforward way to sign the image? For anyone testing our product, it’s challenging to sync, comment out the lines that remove the keys, compile, and then sign the DTB. If there’s an easier method, please share it

Regards,
Shiva Kumar.

closed 2025-11-26 by sathishas89