ARM-software/arm-systemready #474: EFI Partition in systemready WIC Image Uses Microsoft Basic Data GUID Instead of EFI System GUID

smangali-qc, 2025-09-17:
HI @chetan-rathore ,

We observed that in the systemready-dt_acs_live_image.wic image, the first partition intended for EFI boot (BOOT_ACS) is using the GUID:

GUID Code: EBD0A0A2-B9E5-4433-87C0-68B6B72699C7
Type: Microsoft Basic Data
Partition Name: BOOT_ACS
Size: 360 MiB
Start Sector: 2048
End Sector: 739327

This GUID corresponds to Microsoft Basic Data, which is typically used for general-purpose data partitions in Windows environments.
However, for EFI system partitions, the expected GUID is:

EFI System GUID: C12A7328-F81F-11D2-BA4B-00A0C93EC93B

This discrepancy was identified using sgdisk and fdisk tools, and it raises concerns about compatibility and correctness in boot environments, especially when validating against UEFI standards.

our implementation depends on the EFI GUID.

 **Request for Clarification:** 
Whether this GUID assignment is intentional for the EFI partition?
If not, should the image be regenerated with the correct EFI System GUID?
Are there any implications ?

chetan-rathore, 2025-09-18:
Hi @smangali-qc, 

 `Request for Clarification:
Whether this GUID assignment is intentional for the EFI partition?
If not, should the image be regenerated with the correct EFI System GUID?
Are there any implications ?`

Yes, as per requirements of BBR, the system must have a ESP partition. Previously ACS use to carry the first partition as ESP which lead to false positive on systems not having a ESP partition. Hence the ESP partition from the ACS image was removed.

Thanks,
Chetan

chetan-rathore, 2025-09-24:
Hi @smangali-qc, 

If there are no further queries, can this ticket be closed.

chetan-rathore, 2025-09-29:
Hi @smangali-qc, 

If your queries related to this ticket are answered, can we close this ticket.

smangali-qc, 2025-09-29:
HI @chetan-rathore ,

Thanks for the clarification. We’re good to close this ticket.

Regards.
Shiva Kumar

closed 2025-09-29 by chetan-rathore