ARM-software/arm-systemready #475: System ready validation for Lemans-evk

quicAspratap, 2025-09-18:
Issue: The ACS DT validate test which validation of DT bindings will continue to fail until ARM pre-built moves to a newer kernel based on ~6.17.rc7 or 6.18.rc1.

Lemans-evk board is going to be upstream in about 6.17.rc7 or 6.18.rc1
Patches for Lemans-evk board: https://lore.kernel.org/all/175812100719.2051822.15573385822403919775.robh@kernel.org/

Can we switch the validate test to run DT validate test on a newer kernel version to pass the test-case?

/home/root/fdt/fdt: /: failed to match any schema with compatible: ['qcom,lemans-evk', 'qcom,qcs9100', 'qcom,sa8775p']
/home/root/fdt/fdt: /: failed to match any schema with compatible: ['qcom,lemans-evk', 'qcom,qcs9100', 'qcom,sa8775p']

chetan-rathore, 2025-09-24:
Hi @quicAspratap, 

We will be doing the activity to upgrade the linux version which is currently at 6.12 in DT image to higher version. Along with that activity the binding source will also be moved to same version. 

The source of binding documents are at : https://cdn.kernel.org/pub/linux/kernel/v6.x/
I see the latest their at this point is 6.16.x, I am now sure when support for 6.17 or 6.18 will arrive on cdn.kernel.org.

Thanks,
Chetan

quicAspratap, 2025-09-24:
so in that case eg: lemans-evk will be part of 6.18, we will need to wait for 6.18 to be picked in ARM pre-build images to pass the dt-validate test cases?

chetan-rathore, 2025-10-15:
Hello @quicAspratap, 

We have a recent release: [v25.10_DT_3.1.0](https://github.com/ARM-software/arm-systemready/releases/tag/v25.10_DT_3.1.0)
.
The Linux kernel version remains unchanged, but the DT Validate test has been updated to use the 6.16 kernel.

The next DT release is planned for mid-December, which will include a kernel upgrade.
For the DT image, we use the [linux-yocto](https://git.yoctoproject.org/linux-yocto/refs/tags)
The latest available version is 6.16, which will be used for the update candidate.

DT Validate will be upgraded to the latest kernel available around the first week of November.

Thanks,
Chetan

shashim-quic, 2025-10-17:
Hi @chetan-rathore ,

> We have a recent release: [v25.10_DT_3.1.0](https://github.com/ARM-software/arm-systemready/releases/tag/v25.10_DT_3.1.0) . The Linux kernel version remains unchanged, but the DT Validate test has [...]
 
Would be nice if different pieces of s/w update timelines can be documented, so folks can align to it.
If some fixed cadence of updates are followed, can it be published ?

chetan-rathore, 2025-10-17:
Hi @quicAspratap,

- The dt-validate test suite will use kernel 6.18 or later by the end of the first week of November.
- The changes will be merged into the main branch and available in the latest builds by that time.

The next official release is currently planned for mid-December, which will also include a Linux kernel upgrade.

++ @shashim-quic

- The ACS follows two major component upgrade releases per year, typically around Sep/Oct and Mar/Apr.
- Additional releases may occur on a need-basis, outside the regular cadence, depending on requirements.

Thanks for the feedback on publishing the release cadence and upcoming changes.
We’ll look into the best way to make this information more visible.

quicAspratap, 2025-11-28:
@chetan-rathore, lemans-evk is enabled from 6.18-rc7 on cdn.kernel.org.

closed 2025-12-09 by chetan-rathore